DoosanRobotics/doosan-robot2 #180: Robot Disconnecting Unexpectedly with Network Exception Error

gautham-nf, 2025-03-10:
Hello,

I'm encountering an issue while running the dsr_bringup2_moveit with the A0509 robot via the ROS2 package. 

The command I am running
```shell
ros2 launch dsr_bringup2 dsr_bringup2_moveit.launch.py mode:=real model:=a0509 host:=192.168.1.140 color:=blue
```

The robot unexpectedly disconnects during operation, and the following log appears:

```txt
1741590874.0307581 [ros2_control_node-7] [INFO] [1741590874.030586318] [dsr_moveit_controller]: Accepted new action goal
1741590877.7292459 [ros2_control_node-7] [INFO] [1741590877.728812809] [dsr_moveit_controller]: Goal reached, success!
1741591187.7007856 [ros2_control_node-7] Net Exception: Network is unreachable
1741591187.7009828 [ros2_control_node-7] Close UDP Connection
1741591188.9852939 [ros2_control_node-7] Timeout
```

**Steps to Reproduce**:

1. Run the above command to launch the robot using MoveIt.
2. After a period of idle (no movement), the robot unexpectedly disconnects. The disconnection does not occur after a fixed amount of time but happens randomly during idle states.

**Additional Information**:

- The issue occurs intermittently during idle periods.
- I've attached the controller logs for further examination.

[20250309_to_20250310_CLOG.tar.gz](https://github.com/user-attachments/files/19159421/20250309_to_20250310_CLOG.tar.gz)

```log
1741590770.4334762 [INFO] [launch]: All log files can be found below /home/dexus/.ros/log/2025-03-10-12-42-50-432085-NeuralFoundy-14848
1741590770.4335618 [INFO] [launch]: Default logging verbosity is set to INFO
1741590770.5840163 [INFO] [set_config-1]: process started with pid [14852]
1741590770.5842090 [INFO] [run_emulator-2]: process started with pid [14854]
1741590770.5842550 [INFO] [robot_state_publisher-3]: process started with pid [14856]
1741590770.5842874 [INFO] [spawner-4]: process started with pid [14858]
1741590770.5843160 [INFO] [spawner-5]: process started with pid [14860]
1741590770.5843427 [INFO] [spawner-6]: process started with pid [14862]
1741590770.6246696 [robot_state_publisher-3] [INFO] [1741590770.624376571] [robot_state_publisher]: got segment base_link
1741590770.6250231 [robot_state_publisher-3] [INFO] [1741590770.624486749] [robot_state_publisher]: got segment link_1
1741590770.6250951 [robot_state_publisher-3] [INFO] [1741590770.624492019] [robot_state_publisher]: got segment link_2
1741590770.6251402 [robot_state_publisher-3] [INFO] [1741590770.624495265] [robot_state_publisher]: got segment link_3
1741590770.6251802 [robot_state_publisher-3] [INFO] [1741590770.624498100] [robot_state_publisher]: got segment link_4
1741590770.6252182 [robot_state_publisher-3] [INFO] [1741590770.624500876] [robot_state_publisher]: got segment link_5
1741590770.6252534 [robot_state_publisher-3] [INFO] [1741590770.624503511] [robot_state_publisher]: got segment link_6
1741590770.6252909 [robot_state_publisher-3] [INFO] [1741590770.624506015] [robot_state_publisher]: got segment world
1741590770.8637385 [spawner-4] [INFO] [1741590770.863538748] [spawner_dsr_controller2]: waiting for service /controller_manager/list_controllers to become available...
1741590770.8645880 [spawner-5] [INFO] [1741590770.864497371] [spawner_joint_state_broadcaster]: waiting for service /controller_manager/list_controllers to become available...
1741590770.8750725 [spawner-6] [INFO] [1741590770.874951992] [spawner_dsr_moveit_controller]: waiting for service /controller_manager/list_controllers to become available...
1741590770.9276702 [INFO] [set_config-1]: process has finished cleanly [pid 14852]
1741590771.0327218 [INFO] [ros2_control_node-7]: process started with pid [14966]
1741590771.0683663 [ros2_control_node-7] [WARN] [1741590771.068119830] [controller_manager]: [Deprecated] Passing the robot description parameter directly to the control_manager node is deprecated. Use '~/robot_description' topic from 'robot_state_publisher' instead.
1741590771.0686588 [ros2_control_node-7] [INFO] [1741590771.068395301] [resource_manager]: Loading hardware 'a0509' 
1741590771.0755565 [ros2_control_node-7] [INFO] [1741590771.075354031] [resource_manager]: Initialize hardware 'a0509' 
1741590771.0758138 [ros2_control_node-7] [INFO] [1741590771.075632557] [dsr_hw_interface2]: [on_init] Hardware name : a0509, type : system, class type : dsr_hardware2/DRHWInterface
1741590771.0804160 [ros2_control_node-7] [INFO] [1741590771.080309832] [dsr_hw_interface2]: name: 
1741590771.0805926 [ros2_control_node-7] [INFO] [1741590771.080356470] [dsr_hw_interface2]: rate: 100
1741590771.0806313 [ros2_control_node-7] [INFO] [1741590771.080382379] [dsr_hw_interface2]: standby: 5000
1741590771.0806684 [ros2_control_node-7] [INFO] [1741590771.080394683] [dsr_hw_interface2]: command: true
1741590771.0806890 [ros2_control_node-7] [INFO] [1741590771.080401956] [dsr_hw_interface2]: host: 192.168.1.140
1741590771.0807099 [ros2_control_node-7] [INFO] [1741590771.080410803] [dsr_hw_interface2]: port: 12345
1741590771.0807300 [ros2_control_node-7] [INFO] [1741590771.080417937] [dsr_hw_interface2]: mode: real
1741590771.0807490 [ros2_control_node-7] [INFO] [1741590771.080437243] [dsr_hw_interface2]: model: a0509
1741590771.0807672 [ros2_control_node-7] [INFO] [1741590771.080443976] [dsr_hw_interface2]: gripper: none
1741590771.0807855 [ros2_control_node-7] [INFO] [1741590771.080462992] [dsr_hw_interface2]: mobile: none
1741590771.0808034 [ros2_control_node-7] [INFO] [1741590771.080472430] [dsr_hw_interface2]: rt_host: 192.168.137.50
1741590771.0809274 [ros2_control_node-7] [INFO] [1741590771.080523777] [dsr_hw_interface2]: _______________________________________________
1741590771.0809577 [ros2_control_node-7] 
[... 119 lines cut ...]
1741597742.8999624 [rviz2-9] 
1741597742.8999860 [rviz2-9] with this new error message:
1741597742.9000072 [rviz2-9] 
1741597742.9000280 [rviz2-9]   'publisher's context is invalid, at ./src/rcl/publisher.c:389'
1741597742.9000528 [rviz2-9] 
1741597742.9000721 [rviz2-9] rcutils_reset_error() should be called after error handling to avoid this.
1741597742.9000916 [rviz2-9] <<<
1741597742.9005535 [rviz2-9] [INFO] [1741597742.900486189] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Stopping planning scene monitor
1741597742.9028978 [rviz2-9] [INFO] [1741597742.902791860] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Stopping planning scene monitor
1741597742.9036121 [ros2_control_node-7] Close UDP Connection
1741597742.9037430 [ros2_control_node-7] [INFO] [1741597742.903553033] [dsr_hw_interface2]: _______________________________________________
1741597742.9037917 [ros2_control_node-7] 
1741597742.9038155 [ros2_control_node-7] [INFO] [1741597742.903559034] [dsr_hw_interface2]:     CONNECTION IS CLOSED
1741597742.9038351 [ros2_control_node-7] [INFO] [1741597742.903561148] [dsr_hw_interface2]: _______________________________________________
1741597742.9038608 [ros2_control_node-7] 
1741597743.9381447 [ERROR] [move_group-8]: process has died [pid 15091, exit code -11, cmd '/opt/ros/humble/lib/moveit_ros_move_group/move_group --ros-args --params-file /tmp/launch_params_e18l8jfd'].
1741597744.9662809 [ERROR] [rviz2-9]: process has died [pid 15093, exit code -11, cmd '/opt/ros/humble/lib/rviz2/rviz2 -d /home/dexus/ros2_ws/install/dsr_moveit_config_a0509/share/dsr_moveit_config_a0509/launch/moveit.rviz --ros-args -r __node:=rviz2 --params-file /tmp/launch_params_knc8ewdg --params-file /tmp/launch_params_rwpcath3 --params-file /tmp/launch_params_8u56izae --params-file /tmp/launch_params_fo3ukdpm --params-file /tmp/launch_params_mzsnyzzd'].
1741597747.7959752 [ERROR] [ros2_control_node-7]: process[ros2_control_node-7] failed to terminate '5' seconds after receiving 'SIGINT', escalating to 'SIGTERM'
1741597747.7965009 [INFO] [ros2_control_node-7]: sending signal 'SIGTERM' to process[ros2_control_node-7]
1741597747.8035481 [ERROR] [ros2_control_node-7]: process has died [pid 14966, exit code -15, cmd '/opt/ros/humble/lib/controller_manager/ros2_control_node --ros-args -r __ns:=/ --params-file /tmp/launch_params_97kslm9c --params-file /home/dexus/ros2_ws/install/dsr_controller2/share/dsr_controller2/config/dsr_controller2.yaml'].
```

Could you provide insights on what might be causing this issue or suggest any troubleshooting steps?

Thank you!

leeminju531, 2025-03-12:
Hi, @gautham-nf 

I think you need to ensure that the LAN cable is always plugged in on your end.
After the UDP connection is established, the client (ROS process) sends a heartbeat every second. However, the error message `network is unreachable` indicates a failure in sending. I can reproduce the error by deliberately unplugging the LAN cable.

Could you make sure the lan and try again?

Thanks

gautham-nf, 2025-03-12:
Thanks for your response. However, since the robot controller setup is stationary, I don’t expect any LAN disconnections. I have not altered or changed the LAN cable, so I don’t believe that’s the cause of the issue.

leeminju531, 2025-03-24:
@gautham-nf 

I see, I'm about to check this out.
Could you provide the results of` ip route` and` traceroute 192.168.1.140` from your host?

closed 2025-06-05 by leeminju531