DoosanRobotics/doosan-robot2 #194: Inquiry About Setting Up `moveit_servo`

gauthammsam57, 2025-04-02:
Hello,

I have not yet seen the feature or documentation related to setting up moveit_servo. I am wondering if there are any plans for integrating or supporting the `moveit_servo` package for Doosan robots in the near future.

Could you please provide some guidance on whether this setup is planned or if there are any existing alternatives for real-time motion control in this context?

Thank you for your assistance.

leeminju531, 2025-04-03:
As far as I know, MoveIt typically publishes motion commands to the `JointTrajectoryController`, which is loaded during the MoveIt launch. The integration at a low level might already be prepared.
However, we are not explicitly planning to provide guidance or a demo on this at the moment.
If you create one, we’d be happy to see it.

gauthammsam57, 2025-04-29:
Thanks for the response.. Will try this out when I get some free time.

Closing the task for now.

closed 2025-04-29 by gauthammsam57