DoosanRobotics/doosan-robot2 #198: AlterMotion.srv does not work?

Dorianproject, 2025-04-03:
For the A0509 newest Firmware:

$ ros2 service call /dsr01/motion/enable_alter_motion dsr_msgs2/srv/EnableAlterMotion "{n: 10, mode: 0, ref: 0, limit_dpos: [20.0, 20.0], limit_dpos_per: [10.0, 10.0]}"
waiting for service to become available...
requester: making request: dsr_msgs2.srv.EnableAlterMotion_Request(n=10, mode=0, ref=0, limit_dpos=array([20., 20.]), limit_dpos_per=array([10., 10.]))

response:
dsr_msgs2.srv.EnableAlterMotion_Response(success=True)

$ ros2 service call /dsr01/motion/alter_motion dsr_msgs2/srv/AlterMotion "{pos: [0.0, 0.0, -50.0, 0.0, 0.0, 0.0]}"
waiting for service to become available...
requester: making request: dsr_msgs2.srv.AlterMotion_Request(pos=array([  0.,   0., -50.,   0.,   0.,   0.]))

response:
dsr_msgs2.srv.AlterMotion_Response(success=True)

$ ros2 service call /dsr01/system/get_robot_state dsr_msgs2/srv/GetRobotState
waiting for service to become available...
requester: making request: dsr_msgs2.srv.GetRobotState_Request()

response:
dsr_msgs2.srv.GetRobotState_Response(robot_state=1, success=True)

The Robot does not move.

song-ms, 2025-04-09:
Hello @Dorianproject ,

Thank you for sharing the issue. 
Since most of robot logging info is from "bringup launcher file", could you let me know what log in launcher came up by commanding AlterMotion service?

leeminju531, 2025-04-24:
Dear @Dorianproject 

As far as I know, `Alter Motion` doesn't move the robot by itself—it modifies the path while the robot is already in motion.
It would be a good idea to check the manual to confirm whether this is necessary for your use case and to understand how to use it properly.

Also, the `DRL`(in Dart Platform) sample code of the `Doosan_Robotics_Programming_Manual.pdf` might provide better context on how this works.