DoosanRobotics/doosan-robot2 #199: move_spline_joint service doesn't work?

kidpaul94, 2025-04-04:
I have been testing move_spline_joint service in Python environment + virtual robot in Rviz. Below is the code I've written:

```
import rclpy
from rclpy.node import Node
from dsr_msgs2.srv import MoveSplineJoint
from std_msgs.msg import Float64MultiArray

class MoveSplineJointClient(Node):
    def __init__(self):
        super().__init__('move_spline_joint_client')
        self.cli = self.create_client(MoveSplineJoint, '/motion/move_spline_joint')
        while not self.cli.wait_for_service(timeout_sec=1.0):
            self.get_logger().info('Waiting for /motion/move_spline_joint service...')
        self.req = MoveSplineJoint.Request()

    def send_request(self):
        # Define multiple joint positions
        joint_path = [
            [-30., 0.0, 0.0, 0.0, 0.0, 0.0],
            [-60., 0.0, 0.0, 0.0, 0.0, 0.0],
            [-90., 0.0, 0.0, 0.0, 0.0, 0.0],
            [-120., 0.0, 0.0, 0.0, 0.0, 0.0],
            [-150., 0.0, 0.0, 0.0, 0.0, 0.0],
            [-180., 0.0, 0.0, 0.0, 0.0, 0.0],
        ]
        joint_path = np.radians(joint_path).tolist()

        self.req.pos = []
        for joint_pose in joint_path:
            arr = Float64MultiArray()
            arr.data = joint_pose
            self.req.pos.append(arr)

        self.req.vel = np.radians([20.0, 20.0, 20.0, 20.0, 20.0, 20.0]).tolist()
        self.req.acc = np.radians([20.0, 20.0, 20.0, 20.0, 20.0, 20.0]).tolist()
        self.req.pos_cnt = len(joint_path)

def main(args=None):
    rclpy.init(args=args)
    node = MoveSplineJointClient()
    node.send_request()

    node.destroy_node()
    rclpy.shutdown()

if __name__ == '__main__':
    main()
```
However, the robot never moves for some reasons. This problem persists even when I call the service through command prompt directly:

```
ros2 service call /motion/move_spline_joint dsr_msgs2/srv/MoveSplineJoint "pos: [data: [30.0, 30.0, 30.0, 30, 30 ,30], data: [30.0, 30.0, 30.0, 30, 30 ,30]]
pos_cnt: 2
vel:
- 30.0
- 30.0
- 30.0
- 30.0
- 30.0
- 30.0
acc:
- 30.0
- 30.0
- 30.0
- 30.0
- 30.0
- 30.0
time: 0.0
mode: 0
sync_type: 0" 
requester: making request: dsr_msgs2.srv.MoveSplineJoint_Request(pos=[std_msgs.msg.Float64MultiArray(layout=std_msgs.msg.MultiArrayLayout(dim=[], data_offset=0), data=[30.0, 30.0, 30.0, 30.0, 30.0, 30.0]), std_msgs.msg.Float64MultiArray(layout=std_msgs.msg.MultiArrayLayout(dim=[], data_offset=0), data=[30.0, 30.0, 30.0, 30.0, 30.0, 30.0])], pos_cnt=2, vel=array([30., 30., 30., 30., 30., 30.]), acc=array([30., 30., 30., 30., 30., 30.]), time=0.0, mode=0, sync_type=0)

response:
dsr_msgs2.srv.MoveSplineJoint_Response(success=True)
```

kidpaul94, 2025-04-04:
Never mind, the single_robot_simple.py from dsr_example works. I'll close the issue.

closed 2025-04-04 by kidpaul94

Dorianproject, 2025-04-04:
> Never mind, the single_robot_simple.py from dsr_example works. I'll close the issue.

It is important to add the robots name while creating a client, you can see this with 
_ros2 service list_

so to make your code work you have to add /dsr01 

_self.cli = self.create_client(MoveSplineJoint, '/dsr01/motion/move_spline_joint')_

Hope that helps