DoosanRobotics/doosan-robot2 #206: Intermittent error while moving the robot

gauthammsam57, 2025-04-28:
I am getting this error intermittently. I am not able to pin point any pattern where the problem happens. 

```txt
[ros2_control_node-5] #1    Source "/usr/include/c++/11/bits/invoke.h", line 61, in __invoke_impl<void, dsr_controller2::RobotController::on_activate(const rclcpp_lifecycle::State&)::<lambda(std::shared_ptr<dsr_msgs2::srv::GetCurrentPosx_Request_<std::allocator<void> > >, std::shared_ptr<dsr_msgs2::srv::GetCurrentPosx_Response_<std::allocator<void> > >)>&, std::shared_ptr<dsr_msgs2::srv::GetCurrentPosx_Request_<std::allocator<void> > >, std::shared_ptr<dsr_msgs2::srv::GetCurrentPosx_Response_<std::allocator<void> > > > [0x7b3bd324e619]
[ros2_control_node-5]          58:   template<typename _Res, typename _Fn, typename... _Args>
[ros2_control_node-5]          59:     constexpr _Res
[ros2_control_node-5]          60:     __invoke_impl(__invoke_other, _Fn&& __f, _Args&&... __args)
[ros2_control_node-5]       >  61:     { return std::forward<_Fn>(__f)(std::forward<_Args>(__args)...); }
[ros2_control_node-5]          62: 
[ros2_control_node-5]          63:   template<typename _Res, typename _MemFun, typename _Tp, typename... _Args>
[ros2_control_node-5]          64:     constexpr _Res
[ros2_control_node-5] #0    Source "/home/dexus/ros2_ws/src/doosan-robot2/dsr_controller2/src/dsr_controller2.cpp", line 704, in operator() [0x7b3bd31d6e20]
[ros2_control_node-5]         701:     LPROBOT_TASK_POSE cur_posx = Drfl->get_current_posx((COORDINATE_SYSTEM)req->ref);
[ros2_control_node-5]         702:     arr.data.clear();
[ros2_control_node-5]         703:     for (int i = 0; i < NUM_TASK; i++){
[ros2_control_node-5]       > 704:         arr.data.push_back(cur_posx->_fTargetPos[i]);
[ros2_control_node-5]         705:     }
[ros2_control_node-5]         706:     arr.data.push_back(cur_posx->_iTargetSol);
[ros2_control_node-5]         707:     res->task_pos_info.push_back(arr);
[ros2_control_node-5] Segmentation fault (Address not mapped to object [(nil)])
[ERROR] [ros2_control_node-5]: process has died [pid 271235, exit code -11, cmd '/opt/ros/humble/lib/controller_manager/ros2_control_node --ros-args -r __ns:=/dsr01 --params-file /tmp/launch_params_g8lvcim9 --params-file /home/dexus/ros2_ws/install/dsr_controller2/share/dsr_controller2/config/dsr_controller2.yaml'].
```

Full launch logs: [launch.log](https://github.com/user-attachments/files/19939495/launch.log)

I also get this warning on my console: 

[image]

Kindly assist in resolving this error.

Thanks in advance

leeminju531, 2025-04-29:
Segment seems to be come from `GetCurrentPosx` service. Fixed at https://github.com/DoosanRobotics/doosan-robot2/pull/207.
Thanks.

gauthammsam57, 2025-04-29:
Thanks Leeminju for the quick fix. 

But what could be the cause for this error? Cause this happens in the middle of its operation. If it was first I can understand some `init` issue, but this seems to be so random.

leeminju531, 2025-04-29:
Dear @gauthammsam57
 
This issue occurs when calling the `GetCurrentPosx` service, but the previous service doesn't handle failure cases.
I believe your client side might be calling it at certain times.

gauthammsam57, 2025-04-29:
@leeminju531: Got it. But I was curious to why it fails. Does that failure have anything to do with "A heartbeat packet was not received for 5 second" warning.

Thanks for the fix. I guess this issue can be closed from my end once the pull request gets merged.

leeminju531, 2025-05-20:
> Does that failure have anything to do with "A heartbeat packet was not received for 5 second" warning.

Yeah. it's related. but i'm not sure it's the only reason for the failure. if it happens frequently, we should take a closer look at controller.

closed 2025-05-20 by leeminju531