DoosanRobotics/doosan-robot2 #208: Network Timeout and disconnection

gauthammsam57, 2025-04-29:
I started to face this same issue as mentioned in this issue page:  https://github.com/DoosanRobotics/doosan-robot2/issues/87

Abrupt disconnection from the device.

[image]
---
[image]

Could you give some guide lines on how the network should be setup? Currently I can connected to my access point router over ethernet, and my PC over WiFi, to the same router.

song-ms, 2025-04-29:
Hello @gauthammsam57 ,

Logs in your images is that appears when the connection to the robot is lost.
Due to the network environment, the PC is connected to the WiFi router, so the connection may be unstable.
Although it is possible to add code to reconnect when disconnected, the communication will still be unstable, which can cause problems in controlling the robot.
Please check the signal strength between the router and the PC to ensure there are no issues with maintaining the connection.

gauthammsam57, 2025-05-02:
Hello @song-ms,

Thanks for your response. The WiFi router is about 1 meter far from my computer. And signal strength seems to be full. I have connected to 5Ghz network bandwidth. 

I agree it's not always best to have re connection. But after disconnection, there is no way from software to know if the robot moved or not. All of my movement commands get executed and gets success response. I believe that should be changed. 

FYI:> I connected my computer to the Ethernet and it seems to be working well. But for me that's not a permanent option. 

Is there anyway to figure out what went wrong to troubleshoot it at root cause. 

Thank you

leeminju531, 2025-05-22:
@gauthammsam57 ,

> But after disconnection, there is no way from software to know if the robot moved or not. All of my movement commands get executed and gets success response. I believe that should be changed.

Movement commands during disconnection cannot be reliably ensured, and this also involves safety concerns. When movement commands are called synchronously and a disconnection occurs, it could cause the client to block, without receiving a success response. 

> Is there anyway to figure out what went wrong to troubleshoot it at root cause.

The first step would be to ensure the signal strength. If that’s not a permanent option for you,I've added a publisher to detect this case in the system (https://github.com/DoosanRobotics/doosan-robot2/pull/216)