DoosanRobotics/doosan-robot2 #211: Could Not Contact Service dsr01/controller_manager/list_controllers

Jonah1903, 2025-05-05:
I run a ROS2 program to record the robots TCP position over time. Last week this program ran fine, without changing anything I got this error, see picture. 

It initializes, it opens the connection, the servos all click on, the robot lights up white, but then it trows the error that it can't contact the .../list_controllers service. Therefore the joint states don't get published and I can't record the TCP.

[image]

I've power cycled the robot and my PC multiple times, reinstalled ROS2 and DSR, but nothing seems to help.

leeminju531, 2025-05-06:
Dear @Jonah1903 ,

When starting the launch, can you set the `rt_host` parameter as well, not just the `host` parameter?

waledroid, 2025-05-07:
@Jonah1903 i landed on the same error 
where you able to fix it ?  and did starting the  rt_host parameter work ?

Jonah1903, 2025-05-07:
@waledroid I haven’t yet been able to try, will update tomorrow.

Jonah1903, 2025-05-08:
@waledroid It works now, I just added rt_host:=192.168.137.100 to the startup command and it worked. 
Thank you @leeminju531 !!!