DoosanRobotics/doosan-robot2 #220: Connection termination after a single trial of moveit control.

kidpaul94, 2025-06-11:
Hi,

I've consistently faced an issue where my PC keeps losing a connection with the robot. Below is the log after the moveit control:

```
[move_group-7] You can start planning now!
[move_group-7] 
[rviz2-8] [INFO] [1749606738.353090652] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-8] [INFO] [1749606738.353216154] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-8] [INFO] [1749606738.446213082] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-8] Warning: class_loader.impl: SEVERE WARNING!!! A namespace collision has occurred with plugin factory for class rviz_default_plugins::displays::InteractiveMarkerDisplay. New factory will OVERWRITE existing one. This situation occurs when libraries containing plugins are directly linked against an executable (the one running right now generating this message). Please separate plugins out into their own library or just don't link against the library and use either class_loader::ClassLoader/MultiLibraryClassLoader to open.
[rviz2-8]          at line 253 in /opt/ros/humble/include/class_loader/class_loader/class_loader_core.hpp
[rviz2-8] Error:   Semantic description is not specified for the same robot as the URDF
[rviz2-8]          at line 681 in ./src/model.cpp
[rviz2-8] [INFO] [1749606738.566416213] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.0289498 seconds
[rviz2-8] [INFO] [1749606738.566458015] [moveit_robot_model.robot_model]: Loading robot model 'm0609'...
[rviz2-8] [ERROR] [1749606741.826308232] [moveit_ros_visualization.motion_planning_frame]: Action server: /recognize_objects not available
[rviz2-8] [INFO] [1749606741.853266852] [moveit_ros_visualization.motion_planning_frame]: MoveGroup namespace changed: / -> . Reloading params.
[rviz2-8] Error:   Semantic description is not specified for the same robot as the URDF
[rviz2-8]          at line 681 in ./src/model.cpp
[rviz2-8] [INFO] [1749606741.910769828] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.00369257 seconds
[rviz2-8] [INFO] [1749606741.910808295] [moveit_robot_model.robot_model]: Loading robot model 'm0609'...
[rviz2-8] [WARN] [1749606742.012171037] [kdl_parser]: The root link base_link has an inertia specified in the URDF, but KDL does not support a root link with an inertia.  As a workaround, you can add an extra dummy link to your URDF.
[rviz2-8] [INFO] [1749606742.219749781] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting planning scene monitor
[rviz2-8] [INFO] [1749606742.221174822] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/monitored_planning_scene'
[rviz2-8] [INFO] [1749606742.222665844] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting planning scene monitor
[rviz2-8] [INFO] [1749606742.223968400] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/monitored_planning_scene'
[rviz2-8] [INFO] [1749606742.229343727] [interactive_marker_display_98166370760928]: Connected on namespace: /rviz_moveit_motion_planning_display/robot_interaction_interactive_marker_topic
[rviz2-8] [WARN] [1749606742.231518125] [kdl_parser]: The root link base_link has an inertia specified in the URDF, but KDL does not support a root link with an inertia.  As a workaround, you can add an extra dummy link to your URDF.
[rviz2-8] [INFO] [1749606742.234691626] [moveit_ros_visualization.motion_planning_frame]: group manipulator
[rviz2-8] [INFO] [1749606742.234711040] [moveit_ros_visualization.motion_planning_frame]: Constructing new MoveGroup connection for group 'manipulator' in namespace ''
[rviz2-8] [INFO] [1749606742.244060543] [move_group_interface]: Ready to take commands for planning group manipulator.
[rviz2-8] [INFO] [1749606742.252261785] [interactive_marker_display_98166370760928]: Sending request for interactive markers
[rviz2-8] [INFO] [1749606742.283574115] [interactive_marker_display_98166370760928]: Service response received for initialization
[move_group-7] [INFO] [1749606753.199160869] [moveit_move_group_default_capabilities.move_action_capability]: Received request
[rviz2-8] [INFO] [1749606753.199329765] [move_group_interface]: Plan and Execute request accepted
[move_group-7] [INFO] [1749606753.199381520] [moveit_move_group_default_capabilities.move_action_capability]: executing..
[move_group-7] [INFO] [1749606753.231108644] [moveit_move_group_default_capabilities.move_action_capability]: Combined planning and execution request received for MoveGroup action. Forwarding to planning and execution pipeline.
[move_group-7] [INFO] [1749606753.231212258] [moveit_ros.plan_execution]: Planning attempt 1 of at most 1
[move_group-7] [INFO] [1749606753.231228925] [moveit_move_group_capabilities_base.move_group_capability]: Using planning pipeline 'ompl'
[move_group-7] [INFO] [1749606753.232331614] [moveit.ompl_planning.model_based_planning_context]: Planner configuration 'manipulator' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[move_group-7] [INFO] [1749606753.255305720] [moveit.plugins.moveit_simple_controller_manager]: Returned 1 controllers in list
[move_group-7] [INFO] [1749606753.255343737] [moveit.plugins.moveit_simple_controller_manager]: Returned 1 controllers in list
[move_group-7] [INFO] [1749606753.255424325] [moveit_ros.trajectory_execution_manager]: Validating trajectory with allowed_start_tolerance 0.01
[move_group-7] [INFO] [1749606753.331137632] [moveit_ros.trajectory_execution_manager]: Starting trajectory execution ...
[move_group-7] [INFO] [1749606753.331248942] [moveit.plugins.moveit_simple_controller_manager]: Returned 1 controllers in list
[move_group-7] [INFO] [1749606753.331291949] [moveit.plugins.moveit_simple_controller_manager]: Returned 1 controllers in list
[move_group-7] [INFO] [1749606753.331570141] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: sending trajectory to dsr_moveit_controller
[ros2_control_node-6] [INFO] [1749606753.332172771] [dsr_moveit_controller]: Received new action goal
[ros2_control_node-6] [INFO] [1749606753.332231249] [dsr_moveit_controller]: Accepted new action goal
[move_group-7] [INFO] [1749606753.332374301] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: dsr_moveit_controller started execution
[move_group-7] [INFO] [1749606753.332401393] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Goal request accepted!
[ros2_control_node-6] [INFO] [1749606757.730525633] [dsr_moveit_controller]: Goal reached, success!
[move_group-7] [INFO] [1749606757.733221087] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Controller 'dsr_moveit_controller' successfully finished
[ros2_control_node-6] [ERROR] [1749606757.884326667] [dsr_controller2]: [callback OnLogAlarm]
[ros2_control_node-6] [ERROR] [1749606757.884384553] [dsr_controller2]:  level : 3
[ros2_control_node-6] [ERROR] [1749606757.884400723] [dsr_controller2]:  group : 2
[ros2_control_node-6] [ERROR] [1749606757.884408866] [dsr_controller2]:  index : 1908
[ros2_control_node-6] [ERROR] [1749606757.884415816] [dsr_controller2]:  param : 6
[ros2_control_node-6] [ERROR] [1749606757.884422493] [dsr_controller2]:  param : desired position=   -nan[deg], limit=360.000[deg]
[ros2_control_node-6] [ERROR] [1749606757.884432520] [dsr_controller2]:  param : 
[move_group-7] [INFO] [1749606758.031207110] [moveit_ros.trajectory_execution_manager]: Completed trajectory execution with status SUCCEEDED ...
[move_group-7] [INFO] [1749606758.041269723] [moveit_move_group_default_capabilities.move_action_capability]: Solution was found and executed.
[rviz2-8] [INFO] [1749606758.042170076] [move_group_interface]: Plan and Execute request complete!
```

On the teach pendant, I've seen the following error:

```
1.3014
An error occurred during real-time external control.

Param1: A heartbeat packet was not received for 5 seconds.
Param2:
Param3:
```

I would appreciate it your help as this would be a quite critical issue for users who wants to utilize moveit for the robot control.

Note:
I freshly installed the most up-to-date doosan-ros2 package. I didn't modify any part of the original package.

song-ms, 2025-07-08:
Hello @kidpaul94 

Could you please share the version of your controller? 
This information will help us better understand and support your case.

kidpaul94, 2025-07-08:
It is GF03020200.

song-ms, 2025-07-21:
Dear @kidpaul94,

We apologize for the inconvenience caused.
The issue related to servoj has been resolved in software version GF0303000.
Would you kindly update your robot to version 3.3?

closed 2025-08-12 by song-ms

wsbby, 2025-08-12:
Hello,I encountered the same problem“Connection termination after a single trial of moveit control.”So, can this problem be solved by upgrading the version?

[image]
[image]