DoosanRobotics/doosan-robot2 #221: Error while switching between manual and autonomous mode

hvtung26, 2025-06-14:
Hi, I am working with Doosan M1013 and I need to control the robot using c++ between jog() and movel, movesx. So as far as I know, robot need to be in MANUAL_MODE while jogging and AUTONOMOUS MODE while moving with movel and movesx. From AOTONOMOUS TO MANUAL it worked well but while I convert from MANUAL to AUTONOMOUS, the command worked fine but the LED doesnot. It still in blue so 5. 7056 pop up. Could you please help me with my problems. this is my code:
**_while (controller_->Drfl.GetRobotState() != STATE_STANDBY) {
            rclcpp::sleep_for(std::chrono::milliseconds(100));
        }
        controller_->Drfl.stop(STOP_TYPE_SLOW);
        std::this_thread::sleep_for(std::chrono::milliseconds(1000));
        controller_->Drfl.set_robot_mode(ROBOT_MODE_AUTONOMOUS);
        std::this_thread::sleep_for(std::chrono::milliseconds(1000));
        const int check_interval_ms = 500;
        int elapsed = 0;
        while (elapsed < 20000) {
            ROBOT_MODE mode = controller_->Drfl.get_robot_mode();
            if (mode == ROBOT_MODE_AUTONOMOUS) {
                RCLCPP_INFO(this->get_logger(), "Robot mode successfully set.");
                break;
            }
            controller_->Drfl.set_robot_mode(ROBOT_MODE_AUTONOMOUS);
            std::this_thread::sleep_for(std::chrono::milliseconds(check_interval_ms));
            //controller_->Drfl.set_safety_mode(SAFETY_MODE_AUTONOMOUS, SAFETY_MODE_EVENT_MOVE);
            elapsed += check_interval_ms;
        }
        std::this_thread::sleep_for(std::chrono::milliseconds(500));
        bool suc = false;
        while (!suc)
        {
            suc = controller_->Drfl.set_safety_mode(SAFETY_MODE_AUTONOMOUS, SAFETY_MODE_EVENT_MOVE);
            std::this_thread::sleep_for(std::chrono::milliseconds(500));
        }
        RCLCPP_INFO(this->get_logger(), "Safety mode set to AUTONOMOUS with EVENT_MOVE");
        std::this_thread::sleep_for(std::chrono::milliseconds(2000));
        if (sync_type == 0)
        {
            result = controller_->Drfl.movel(pos, vel, acc, time, (MOVE_MODE)mode, (MOVE_REFERENCE)ref, radius, (BLENDING_SPEED_TYPE)blend_type);
    
        }_**

RCLCPP_INFO(this->get_logger(), "Robot mode successfully set."); and RCLCPP_INFO(this->get_logger(), "Safety mode set to AUTONOMOUS with EVENT_MOVE"); debug well at terminal but LED doesnot change its color.

song-ms, 2025-06-18:
Hello @hvtung26 ,

if you use the commnad of "Drfl.set_safety_mode(SAFETY_MODE_AUTONOMOUS, SAFETY_MODE_EVENT_MOVE)", you have to close the SAFETY_MODE_EVENT_MOVE command by  "Drfl.set_safety_mode(SAFETY_MODE_AUTONOMOUS, SAFETY_MODE_EVENT_STOP)" before switching MANUAL_MODE .
if you use "movel, movesx", I recommand to just use Drfl.set_robot_mode(ROBOT_MODE_AUTONOMOUS). not set_safety_mode.

song-ms, 2025-07-16:
Hi @hvtung26 

It seems like comunication issue between robot and api.
Could you contact our CS team by using below url ? 
https://robotics.doosan.com/hc/en-us

when you write the inquiry, it will be helpful to share the robot log file(not ros). 

Thank you