DoosanRobotics/doosan-robot2 #222: Real-time control with Doosan M1013 – Setup notes and issue with `rt_shutdown` requirement

will-44, 2025-06-17:
Hi,

I'm using a **Doosan M1013** and wanted to control it in **real-time** using the [dsr\_realtime\_control example](https://github.com/DoosanRobotics/doosan-robot2/tree/humble/dsr_example2/dsr_realtime_control). I wanted to share my setup and experience, which might help others—and also raise a question about a behavior I observed.

## My Setup:

* I'm running **ROS 2 in a Docker container** with real-time permissions enabled, following the [ros2\_control documentation](https://control.ros.org/master/doc/ros2_control/controller_manager/doc/userdoc.html).
* The **robot and PC are connected via Ethernet** (not Wi-Fi), since real-time communication runs at 3 kHz (0.333 ms), and Wi-Fi introduces noticeable jitter.
* I had to change the robot's IP address to fit the `192.168.50.xxx` subnet. **Important note:** the real-time communication listens on `192.168.xxx.50`, so do *not* assign `.50` to the robot itself.

## Running the Example:

To run the **gravity compensation example**, there are two options as described in the README [tutorial](https://github.com/DoosanRobotics/doosan-robot2/tree/humble/dsr_example2/dsr_realtime_control#tutorial). In both cases:

* If the IP address is changed, make sure to update:

  * `rt_init.cpp` or `realtime_api.cpp`
  * The `host` and `rt_host` in the `dsr_bringup2` launch

### My issue – `rt_shutdown` Required to Avoid Issues

Here's the part where I could use some clarification. When following the steps in the README [tutorial](https://github.com/DoosanRobotics/doosan-robot2/tree/humble/dsr_example2/dsr_realtime_control#tutorial):
* **Option 1:** nothing happens after launching the example
* **Option 2:** the `dsr_bringup2` node crashes

After some trial and error, I found that **for both options**, you need to run the following command *after* launching `dsr_bringup2`:

```bash
ros2 run dsr_realtime_control rt_shutdown
```

This command terminates the existing UDP communication, allowing a new real-time session to be established properly. Without this step, the example either fails silently or crashes, depending on the option used.

After that, the exemple will work perfectly !

---

If anyone has insights into why `rt_shutdown` is required or if there's a cleaner way to handle this, I'd appreciate your thoughts. Happy to answer questions if anyone is trying a similar setup! (my final goal is to wrap up in a launch file)

Thanks!

song-ms, 2025-07-08:
Hi @will-44 

The examples you mentioned, including the bringup process, do contain control-related code following the RT open connection. However, since only one client is permitted to enable control at a time, attempting to do so from multiple nodes may cause the connection to be interrupted. This is likely why it behaves as expected only after rt_shutdown, as you pointed out.
We fully acknowledge that this aspect should be taken into account in the RT examples, and we will update them accordingly to reflect this consideration.

If you are implementing it via a launch file, you may skip the RT connection step after bringup and directly proceed with the RT-related motion commands.

song-ms, 2025-07-21:
Hello @will-44 ,

The issue has been addressed and the fix has been uploaded. 
Kindly review it, and if you encounter any additional problems, please feel free to reopen the issue. 

Thank you