DoosanRobotics/doosan-robot2 #233: gazebo: Requesting list of world names

PARKBONG, 2025-07-15:
command:
`ros2 launch dsr_bringup2 dsr_bringup2_gazebo.launch.py mode:=virtual host:=127.0.0.1 port:=12346 name:=dsr01 x:=0 y:=0`

After the command Gazebo, windows pop up, but nothing shows. 

```
[ros2_control_node-6] [INFO] [1752564240.782425653] [dsr_hw_interface2]: _______________________________________________
[ros2_control_node-6] 
[ros2_control_node-6] [INFO] [1752564240.782429839] [dsr_hw_interface2]:     Emulator Mode
[ros2_control_node-6] [INFO] [1752564240.782751995] [dsr_hw_interface2]:     DRCF version = GF03020000
[ros2_control_node-6] [INFO] [1752564240.782771407] [dsr_hw_interface2]:     DRFL version = GL013202
[ros2_control_node-6] [INFO] [1752564240.782776612] [dsr_hw_interface2]:     m_nVersionDRCF = 3020000
[ros2_control_node-6] [INFO] [1752564240.782780835] [dsr_hw_interface2]: _______________________________________________

...

[ign gazebo-7] [Wrn] [Gui.cc:261] Waited for 10s for a subscriber to [/gazebo/starting_world] and got none.
[ign gazebo-7] [GUI] [Wrn] [Application.cc:797] [QT] qrc:/qml/StyleDialog.qml:112:3: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
[ign gazebo-7] [GUI] [Wrn] [Application.cc:797] [QT] qrc:/qml/StyleDialog.qml:105:3: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
[ign gazebo-7] [GUI] [Wrn] [Application.cc:797] [QT] qrc:/qml/StyleDialog.qml:98:3: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
[ign gazebo-7] [GUI] [Wrn] [Application.cc:797] [QT] qrc:qml/Main.qml:102:3: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
[ign gazebo-7] [GUI] [Wrn] [Application.cc:797] [QT] qrc:/qml/PluginMenu.qml:27:3: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
[ign gazebo-7] [GUI] [Wrn] [Application.cc:797] [QT] file::/Gazebo/GazeboDrawer.qml:242:3: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }

[create-9] [INFO] [1752564128.152204704] [dsr01.gz.ros_gz_sim]: Requesting list of world names.
[create-9] [INFO] [1752564133.152452258] [dsr01.gz.ros_gz_sim]: Requesting list of world names.
[spawner-10] [WARN] [1752564135.403737730] [dsr01.gz.spawner_dsr_position_controller]: Could not contact service /dsr01/gz/controller_manager/list_controllers
[spawner-10] [INFO] [1752564135.403915928] [dsr01.gz.spawner_dsr_position_controller]: waiting for service /dsr01/gz/controller_manager/list_controllers to become available...
```

PARKBONG, 2025-08-04:
Same Issue occurs,
I share the entire terminal logs
```

user@pc: ros2 launch dsr_bringup2 dsr_bringup2_gazebo.launch.py mode:=virtual host:=127.0.0.1 port:=12346 model:=m1509 name:=dsr01 x:=0 y:=0'
[INFO] [launch]: All log files can be found below /home/sms/.ros/log/2025-08-04-02-33-17-170170-sms-23060
[INFO] [launch]: Default logging verbosity is set to INFO
[WARNING] [launch_ros.actions.node]: Parameter file path is not a file: /home/sms/ros2_ws/install/dsr_controller2/share/dsr_controller2/config/joint_state_broadcaster.yaml
[INFO] [run_emulator-1]: process started with pid [23063]
[INFO] [gazebo_connection-2]: process started with pid [23065]
[INFO] [robot_state_publisher-3]: process started with pid [23067]
[INFO] [spawner-4]: process started with pid [23069]
[INFO] [spawner-5]: process started with pid [23071]
[INFO] [ros2_control_node-6]: process started with pid [23073]
[ros2_control_node-6] [WARN] [1754289197.392978458] [dsr01.controller_manager]: [Deprecated] Passing the robot description parameter directly to the control_manager node is deprecated. Use '~/robot_description' topic from 'robot_state_publisher' instead.
[ros2_control_node-6] [INFO] [1754289197.393291659] [resource_manager]: Loading hardware 'm1509' 
[robot_state_publisher-3] Warning: link 'world' material 'gray' undefined.
[robot_state_publisher-3]          at line 84 in ./urdf_parser/src/model.cpp
[robot_state_publisher-3] Warning: link 'world' material 'gray' undefined.
[robot_state_publisher-3]          at line 84 in ./urdf_parser/src/model.cpp
[robot_state_publisher-3] [WARN] [1754289197.398298046] [kdl_parser]: The root link world has an inertia specified in the URDF, but KDL does not support a root link with an inertia.  As a workaround, you can add an extra dummy link to your URDF.
[robot_state_publisher-3] [INFO] [1754289197.398368829] [dsr01.robot_state_publisher]: got segment base_link
[robot_state_publisher-3] [INFO] [1754289197.398418764] [dsr01.robot_state_publisher]: got segment link_1
[robot_state_publisher-3] [INFO] [1754289197.398422035] [dsr01.robot_state_publisher]: got segment link_2
[robot_state_publisher-3] [INFO] [1754289197.398424273] [dsr01.robot_state_publisher]: got segment link_3
[robot_state_publisher-3] [INFO] [1754289197.398426436] [dsr01.robot_state_publisher]: got segment link_4
[robot_state_publisher-3] [INFO] [1754289197.398428446] [dsr01.robot_state_publisher]: got segment link_5
[robot_state_publisher-3] [INFO] [1754289197.398430541] [dsr01.robot_state_publisher]: got segment link_6
[robot_state_publisher-3] [INFO] [1754289197.398432513] [dsr01.robot_state_publisher]: got segment world
[ros2_control_node-6] [INFO] [1754289197.400107854] [resource_manager]: Initialize hardware 'm1509' 
[ros2_control_node-6] [INFO] [1754289197.400372012] [dsr_hw_interface2]: model : m1509
[ros2_control_node-6] [INFO] [1754289197.400385509] [dsr_hw_interface2]: mode : virtual
[ros2_control_node-6] [INFO] [1754289197.400391102] [dsr_hw_interface2]: port : 12346
[ros2_control_node-6] [INFO] [1754289197.400402608] [dsr_hw_interface2]: rt_host : 192.168.137.50
[ros2_control_node-6] [INFO] [1754289197.400407498] [dsr_hw_interface2]: host : 127.0.0.1
[ros2_control_node-6] [INFO] [1754289197.400461281] [dsr_hw_interface2]: [on_init] Hardware name : m1509, type : system, class type : dsr_hardware2/DRHWInterface
[ros2_control_node-6] [INFO] [1754289197.400539776] [dsr_hw_interface2]: _______________________________________________
[ros2_control_node-6] 
[ros2_control_node-6] [INFO] [1754289197.400545633] [dsr_hw_interface2]:     INITAILIZE
[ros2_control_node-6] [INFO] [1754289197.400549675] [dsr_hw_interface2]: _______________________________________________
[ros2_control_node-6] 
[ros2_control_node-6] [INFO] [1754289197.400797050] [dsr_hw_interface2]: Connecting failure.. retry...
[run_emulator-1] Run Emulator of the Doosan Robot Controller
[run_emulator-1] dirname: /home/sms/ros2_ws/install/dsr_common2/share/dsr_common2/bin/run_drcf.sh
[run_emulator-1] server_port: 12346
[run_emulator-1] robot model: M1509 
[run_emulator-1] ns: dsr01
[run_emulator-1] run : 'dsr01_emulator' ...
[spawner-4] [INFO] [1754289197.549712516] [dsr01.spawner_dsr_controller2]: waiting for service /dsr01/controller_manager/list_controllers to become available...
[spawner-5] [INFO] [1754289197.551645917] [dsr01.spawner_joint_state_broadcaster]: waiting for service /dsr01/controller_manager/list_controllers to become available...
[run_emulator-1] 38f92f0e7df76a63ea9fec26d1aaaecd62fd066369eff540e85c136bab805031
[run_emulator-1] ARCH: 64-bit
[ros2_control_node-6] [INFO] [1754289198.001692793] [dsr_hw_interface2]: Connecting failure.. retry...
[ros2_control_node-6] [INFO] [1754289198.603752900] [dsr_hw_interface2]: Connected to DRCF
[ros2_control_node-6] [INFO] [1754289198.603814319] [dsr_hw_interface2]: INITIAL MANAGE_ACCESS_CONTROL_FORCE_REQUEST called
[ros2_control_node-6] [INFO] [1754289198.603956319] [dsr_hw_interface2]: ROBOT_STATE : STATE_SAFE_OFF
[ros2_control_node-6] [INFO] [1754289198.604384854] [dsr_hw_interface2]: AUTHORITY : MONITORING_ACCESS_CONTROL_GRANT
[ros2_control_node-6] [INFO] [1754289198.604387463] [dsr_hw_interface2]: INITIAL AUTHORITY GRANTED !!!
[ros2_control_node-6] [INFO] [1754289198.604640258] [dsr_hw_interface2]: ROBOT_STATE : STATE_SAFE_OFF
[ros2_control_node-6] [INFO] [1754289199.604420687] [dsr_hw_interface2]: INITIAL CONTROL_SERVO_ON called
[... cut ...]
[create-9] [INFO] [1754289236.815669530] [dsr01.gz.ros_gz_sim]: Requesting list of world names.
[create-9] [INFO] [1754289241.815910482] [dsr01.gz.ros_gz_sim]: Requesting list of world names.
[spawner-10] [WARN] [1754289244.350411114] [dsr01.gz.spawner_dsr_position_controller]: Could not contact service /dsr01/gz/controller_manager/list_controllers
[spawner-10] [INFO] [1754289244.350613068] [dsr01.gz.spawner_dsr_position_controller]: waiting for service /dsr01/gz/controller_manager/list_controllers to become available...

......

```

PARKBONG, 2025-08-04:
And here is the version info. 

$ gazebo --version
Gazebo multi-robot simulator, version 11.10.2
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

$ ign gazebo --version
Gazebo Sim, version 6.17.0
Copyright (C) 2018 Open Source Robotics Foundation.
Released under the Apache 2.0 License.
_______________________________________________
[ros2_control_node-6] 
[ros2_control_node-6] [INFO] [1754289200.604667767] [dsr_hw_interface2]:     Emulator Mode
[ros2_control_node-6] [INFO] [1754289200.605031723] [dsr_hw_interface2]:     DRCF version = GF03020000
[ros2_control_node-6] [INFO] [1754289200.605050430] [dsr_hw_interface2]:     DRFL version = GL013300
[ros2_control_node-6] [INFO] [1754289200.605055365] [dsr_hw_interface2]:     m_nVersionDRCF = 3020000
[ros2_control_node-6] [INFO] [1754289200.605059303] [dsr_hw_interface2]: _______________________________________________

song-ms, 2025-08-12:
Hello @PARKBONG ,

It seems that the GUI of ros_gz_sim is not opening properly.
Could you check if the GUI works correctly when you test it using the command:

ros2 launch ros_gz_sim gz_sim.launch.py

PARKBONG, 2025-08-12:
[image]

Of course, it works. 

[image]

And the virtual mode is still not working.

song-ms, 2025-08-12:
@PARKBONG 

Could you try opening an empty GUI in Gazebo Quick Start?
When I tested it on our two internal PCs, it worked fine as below

```
ms@ms ~/ros2_ws $ ros2 launch dsr_bringup2 dsr_bringup2_gazebo.launch.py mode:=virtual host:=127.0.0.1 port:=12346 model:=m1509 name:=dsr01 x:=0 y:=0
[INFO] [launch]: All log files can be found below /home/ms/.ros/log/2025-08-12-15-41-50-758753-ms-177169
[INFO] [launch]: Default logging verbosity is set to INFO
[WARNING] [launch_ros.actions.node]: Parameter file path is not a file: /home/ms/ros2_ws/install/dsr_controller2/share/dsr_controller2/config/joint_state_broadcaster.yaml
[INFO] [run_emulator-1]: process started with pid [177182]
[INFO] [gazebo_connection-2]: process started with pid [177184]
[INFO] [robot_state_publisher-3]: process started with pid [177186]
[INFO] [spawner-4]: process started with pid [177188]
[INFO] [spawner-5]: process started with pid [177190]
[INFO] [ros2_control_node-6]: process started with pid [177192]
[ros2_control_node-6] [WARN] [1754980911.038763698] [dsr01.controller_manager]: [Deprecated] Passing the robot description parameter directly to the control_manager node is deprecated. Use '~/robot_description' topic from 'robot_state_publisher' instead.
[ros2_control_node-6] [INFO] [1754980911.038987425] [resource_manager]: Loading hardware 'm1509' 
[ros2_control_node-6] [INFO] [1754980911.044217853] [resource_manager]: Initialize hardware 'm1509' 
[ros2_control_node-6] [INFO] [1754980911.044356847] [dsr_hw_interface2]: model : m1509
[ros2_control_node-6] [INFO] [1754980911.044362729] [dsr_hw_interface2]: mode : virtual
[ros2_control_node-6] [INFO] [1754980911.044365282] [dsr_hw_interface2]: port : 12346
[ros2_control_node-6] [INFO] [1754980911.044370468] [dsr_hw_interface2]: rt_host : 192.168.137.50
[ros2_control_node-6] [INFO] [1754980911.044372506] [dsr_hw_interface2]: host : 127.0.0.1
[ros2_control_node-6] [INFO] [1754980911.044397657] [dsr_hw_interface2]: [on_init] Hardware name : m1509, type : system, class type : dsr_hardware2/DRHWInterface
[ros2_control_node-6] [INFO] [1754980911.044440458] [dsr_hw_interface2]: _______________________________________________
[ros2_control_node-6] 
[ros2_control_node-6] [INFO] [1754980911.044442789] [dsr_hw_interface2]:     INITAILIZE
[ros2_control_node-6] [INFO] [1754980911.044444508] [dsr_hw_interface2]: _______________________________________________
[ros2_control_node-6] 
[ros2_control_node-6] [INFO] [1754980911.044595715] [dsr_hw_interface2]: Connecting failure.. retry...
[robot_state_publisher-3] [INFO] [1754980911.044806101] [dsr01.robot_state_publisher]: got segment base_link
[robot_state_publisher-3] [INFO] [1754980911.044919096] [dsr01.robot_state_publisher]: got segment link_1
[robot_state_publisher-3] [INFO] [1754980911.044931617] [dsr01.robot_state_publisher]: got segment link_2
[robot_state_publisher-3] [INFO] [1754980911.044939399] [dsr01.robot_state_publisher]: got segment link_3
[robot_state_publisher-3] [INFO] [1754980911.044946340] [dsr01.robot_state_publisher]: got segment link_4
[robot_state_publisher-3] [INFO] [1754980911.044952960] [dsr01.robot_state_publisher]: got segment link_5
[robot_state_publisher-3] [INFO] [1754980911.044959301] [dsr01.robot_state_publisher]: got segment link_6
[robot_state_publisher-3] [INFO] [1754980911.044965468] [dsr01.robot_state_publisher]: got segment world
[run_emulator-1] Run Emulator of the Doosan Robot Controller
[run_emulator-1] dirname: /home/ms/ros2_ws/install/dsr_common2/share/dsr_common2/bin/run_drcf.sh
[run_emulator-1] server_port: 12346
[run_emulator-1] robot model: M1509 
[run_emulator-1] ns: dsr01
[run_emulator-1] run : 'dsr01_emulator' ...
[spawner-5] [INFO] [1754980911.225412873] [dsr01.spawner_joint_state_broadcaster]: waiting for service /dsr01/controller_manager/list_controllers to become available...
[spawner-4] [INFO] [1754980911.267811056] [dsr01.spawner_dsr_controller2]: waiting for service /dsr01/controller_manager/list_controllers to become available...
[run_emulator-1] 8160af879d2a19b77f8656c6e1c076c93215be87d34e09aefff057d4f49ca0e0
[run_emulator-1] ARCH: 64-bit
[ros2_control_node-6] [INFO] [1754980911.645556194] [dsr_hw_interface2]: Connecting failure.. retry...
[ros2_control_node-6] [INFO] [1754980912.247907498] [dsr_hw_interface2]: Connected to DRCF
[ros2_control_node-6] [INFO] [1754980912.247983559] [dsr_hw_interface2]: INITIAL MANAGE_ACCESS_CONTROL_FORCE_REQUEST called
[ros2_control_node-6] [INFO] [1754980912.248181044] [dsr_hw_interface2]: ROBOT_STATE : STATE_SAFE_OFF
[ros2_control_node-6] [INFO] [1754980912.249002824] [dsr_hw_interface2]: AUTHORITY : MONITORING_ACCESS_CONTROL_GRANT
[ros2_control_node-6] [INFO] [1754980912.249013629] [dsr_hw_interface2]: INITIAL AUTHORITY GRANTED !!!
[ros2_control_node-6] [INFO] [1754980912.249295165] [dsr_hw_interface2]: ROBOT_STATE : STATE_SAFE_OFF
[ros2_control_node-6] [INFO] [1754980913.248399968] [dsr_hw_interface2]: INITIAL CONTROL_SERVO_ON called
[ros2_control_node-6] [INFO] [1754980913.248794899] [dsr_hw_interface2]: ROBOT_STATE : STATE_STANDBY
[ros2_control_node-6] [INFO] [1754980913.248806415] [dsr_hw_interface2]: INITIAL STATE_STANDBY !!!
[ros2_control_node-6] [INFO] [1754980914.248557855] [dsr_hw_interface2]: _______________________________________________
[ros2_control_node-6] 
[ros2_control_node-6] [INFO] [1754980914.248613586] [dsr_hw_interface2]:     OPEN CONNECTION
[... cut ...]
[spawner-10] [INFO] [1754980918.412215034] [dsr01.gz.spawner_dsr_position_controller]: Configured and activated dsr_position_controller
[INFO] [spawner-10]: process has finished cleanly [pid 177752]
[ruby $(which ign) gazebo-7] libEGL warning: egl: failed to create dri2 screen
[ruby $(which ign) gazebo-7] libEGL warning: egl: failed to create dri2 screen
[ruby $(which ign) gazebo-7] [GUI] [Wrn] [Component.hh:189] Trying to deserialize component with data type [N3sdf3v125WorldE], which doesn't have `operator>>`. Component will not be deserialized.
```

closed 2025-09-08 by song-ms