DoosanRobotics/doosan-robot2 #239: Maintaning constant setted force while motion

Bala1411, 2025-07-23:
Hi,
I’m working with the Doosan a0509s robot,using ros2 ,python and I’m trying to implement force compliance during a predefined motion path.
I want the robot to follow a predefined path, and maintain a specified force (on the Z-axis) during motion.
If any external forces or obstacles are encountered, the robot should continue its motion with compliance — adjusting the trajectory slightly while still maintaining the target force.
I have used the check_force_condition(), task_complaince_ctrl and set_desired_force().

I can't get correctly how to use those commands in an order.

Can anyone suggest me a guide to use those commands to maintain the setted force while motion?

Bala1411, 2025-07-23:
while rclpy.ok():
                movej(readypos, vel=40, acc=40)  
                task_compliance_ctrl()
                set_desired_force(fd=[0, 0, -20, 0, 0, 0], dir=[0, 0, 1, 0, 0, 0], mod=DR_FC_MOD_REL)
                wait(0.5)
                while not check_force_condition(DR_AXIS_Z, max=15):
                        movel(posx(-100, 0, 0, 0, 0, 0), mod=DR_FC_MOD_REL)
                        # pass
                release_force()
                release_compliance_ctrl()
                movej(readypos, vel=40, acc=40)
rclpy.shutdown()
This is how I have used the related commands for force compliance segment.

song-ms, 2025-07-30:
Hello @Bala1411 ,

I will forward this to the relevant team and get back to you as soon as I receive their response.