DoosanRobotics/doosan-robot2 #240: Heartbeat packet missing error

Bala1411, 2025-07-26:
@doosan-robotics @leeminju531 @song-ms 
Hi,
Whenever I disconnect from the ros2 control I got an error as heartbeat packet missing for 5 seconds in both terminal and Teachpendant. Could you kindly explain why it happens and how to solve this issue. I am getting this error each and every time when disconnected.
Here below I will attach the images of error.

[image]

[image]

song-ms, 2025-07-30:
Hello @Bala1411 

This issue appears to have occurred because the realtime control connection was not cleanly terminated when the connection was lost.
To address this, we have updated the relevant code and will share the modified file for your review.

Bala1411, 2025-07-30:
@song-ms 
Thank you for the response.