DoosanRobotics/doosan-robot2 #250: Is the "Direct Teaching" feature currently supported?

sukhyun0721, 2025-08-05:
Hello,

I'm writing to inquire about the "direct teaching" feature.

I saw in an older issue (#6) that it was mentioned the direct teaching feature would be added in the next API.

Could you please confirm if this feature is supported in the current version?

We need this functionality for a software we are developing. If it is not available, could you please let us know if there are any alternative solutions?

Thank you.

song-ms, 2025-08-12:
Hello @sukhyun0721 ,

Currently, "direct teaching" can be activated with pressing cockpit button in Manual mode.

song-ms, 2025-08-13:
I’m closing this issue. If anything else comes up, please open a new issue.

closed 2025-08-13 by song-ms