DoosanRobotics/doosan-robot2 #251: Issue with `/tcp/set_current_tcp` service returning `success=False` when setting Gripper TCP

gauthammsam57, 2025-08-11:
## Description:

I'm currently working with the `doosan-robot2` ROS2 package and trying to use the service `/dsr01/tcp/set_current_tcp` to set the TCP to `Gripper`. 

When sending the following request:

```python
requester: making request: dsr_msgs2.srv.SetCurrentTcp_Request(name='Gripper')

response:
dsr_msgs2.srv.SetCurrentTcp_Response(success=False)
```

I've already configured Gripper and Cam_Gripper in the TCP settings in the Doosan console, but when sending the request to set the TCP to Gripper, it fails with success=False.

In the main launch file, I can see the following error messages in the log:

```txt
[ros2_control_node-5] [WARN] [1754914253.447311702] [dsr_hw_interface2]: [callback OnLogAlarm]
[ros2_control_node-5] [WARN] [1754914253.447330122] [dsr_hw_interface2]:  level : 2
[ros2_control_node-5] [WARN] [1754914253.447332862] [dsr_hw_interface2]:  group : 5
[ros2_control_node-5] [WARN] [1754914253.447334692] [dsr_hw_interface2]:  index : 7170
[ros2_control_node-5] [WARN] [1754914253.447336512] [dsr_hw_interface2]:  param : 1
[ros2_control_node-5] [WARN] [1754914253.447338432] [dsr_hw_interface2]:  param : 
[ros2_control_node-5] [WARN] [1754914253.447340092] [dsr_hw_interface2]:  param : 
[ros2_control_node-5] [WARN] [1754914253.447900083] [dsr_hw_interface2]: [callback OnLogAlarm]
[ros2_control_node-5] [WARN] [1754914253.447906693] [dsr_hw_interface2]:  level : 2
[ros2_control_node-5] [WARN] [1754914253.447908943] [dsr_hw_interface2]:  group : 5
[ros2_control_node-5] [WARN] [1754914253.447910643] [dsr_hw_interface2]:  index : 7170
[ros2_control_node-5] [WARN] [1754914253.447912303] [dsr_hw_interface2]:  param : 0
[ros2_control_node-5] [WARN] [1754914253.447913983] [dsr_hw_interface2]:  param : 
[ros2_control_node-5] [WARN] [1754914253.447915583] [dsr_hw_interface2]:  param : 
```

I am currently using the latest commit from the humble-devel branch. Can anyone help identify why this issue is occurring or if additional configuration is required?

## Steps to Reproduce:
Launch the ROS2 node with the Doosan robot.
```bash
ros2 launch dsr_bringup2 dsr_bringup2_rviz.launch.py mode:=real host:=1.1.1.3 model:=a0912 color:=blue
```
Send a request to the /dsr01/tcp/set_current_tcp service with name='Gripper'.

Observe the response is success=False.

## Expected Behavior:
The service should return success=True when setting the Gripper TCP, as it is properly configured in the Doosan console.

## Additional Information:
ROS2 Version: Humble
Doosan ROS2 package version: Latest from humble-devel branch
Robot model: a0509

Any help would be greatly appreciated!

gauthammsam57, 2025-08-12:
Thanks @song-ms for the response. 

Could you kindly guide on how to set the current tcp?
From what I know when I connect over ROS launch it automatically does to auto mode. How then do I change to manual and then set the TCP?

song-ms, 2025-08-12:
@gauthammsam57 

Here’s a clear, professional way to say it:

To switch the robot to Manual mode, run:

```
ros2 service call /dsr01/system/set_robot_mode dsr_msgs2/srv/SetRobotMode "robot_mode: 0"
```
Please complete your TCP setup in Manual mode.

Before executing any motion, you must switch back to Auto mode:
```
ros2 service call /dsr01/system/set_robot_mode dsr_msgs2/srv/SetRobotMode "robot_mode: 1"
```
This ensures motions run properly and safely.

gauthammsam57, 2025-08-12:
Got it. This setup worked. 

Thanks Song.

closed 2025-08-12 by gauthammsam57

Bala1411, 2025-09-04:
Hello @gauthammsam57 

I have reviewed your issue, and I’ve faced a similar one myself. To resolve it, you need to add a motion command before calling the manual mode setting service, which switches the robot from automatic to manual mode. This approach has a higher success rate compared to directly calling the manual mode setting service. Within the motion command, you only need to apply a very small change in any of the axes.

Steps I followed:

Ensure the robot is in Autonomous mode.

Apply a small motion (e.g., 0.5° or 1°).

Call the manual mode setting service.

Call the TCP setting service.

I am not completely sure if this is the “correct” approach, but this method worked for me. Please give it a try and let me know if it resolves your issue as well.