DoosanRobotics/doosan-robot2 #267: move_pause and resume correct implementation?

Dorianproject, 2025-08-26:
When calling the move_pause service, the robot not only stops moving but also deactivates its servos.
This is unexpected, since in the past the service worked as intended (pause motion while keeping the servos active).

It seems like this is a regression or an unintended change.

Expected Behavior
- Motion should pause.
- Servos should remain active so that motion can be resumed with move_resume.

Actual Behavior
- Motion pauses.
- Servos are also deactivated.

move_resume cannot resume motion as intended.

```
    def motion_pause(self):
        """Pause running motion asynchron"""
        client, req = self._create_client_and_request(
            MovePause, '/dsr01/motion/move_pause', {}
        )
        return client.call_async(req) if client else None

    def motion_resume(self):
        """resume paused motion asynchron"""
        client, req = self._create_client_and_request(
            MoveResume, '/dsr01/motion/move_resume', {}
        )
        return client.call_async(req) if client else None
```

yurirocha15, 2025-09-03:
Hello @Dorianproject,

Could you please share the following information:
- Controller version (you can find version info when starting a launch file)
- ROS version
- Robot model

If possible, could you also share the terminal log when calling the services?

Thank you.

yurirocha15, 2025-11-27:
@Dorianproject, sorry for the delay.

I tested the current humble code, and both move_pause and move_resume are working.

It is important to note that to call other services while moving, the motion call should be called in async mode (`request.sync_type = 1`). If you use the default sync_type value (sync mode), the API will wait until the motion ends before returning the answer to ROS. This means that any other service call would be "stuck" in the API queue until the current motion finishes.

### Example call

- terminal 1: `ros2 launch dsr_bringup2 dsr_bringup2_rviz.launch <params>`
- terminal 2: 
  - `ros2 service call /dsr01/motion/move_joint dsr_msgs2/srv/MoveJoint '{"pos": <pose>, "time": 10.0, "sync_type": 1}'` 
  - `ros2 service call /dsr01/motion/move_pause dsr_msgs2/srv/MovePause '{}'`
  - `ros2 service call /dsr01/motion/move_resume dsr_msgs2/srv/MoveResume '{}'`

Can you confirm that you are using sync_type = 1?

Thank you.

yurirocha15, 2025-12-02:
By `robot_state_publisher`, do you mean the node that updates the TF? Most of the launch files in the `dsr_bringup2` package also spawn a `robot_state_publisher`. (For example: https://github.com/DoosanRobotics/doosan-robot2/blob/humble/dsr_bringup2/launch/dsr_bringup2_rviz.launch.py#L126). If you are using your own custom launch logic, then you would need to create the `robot_state_publisher` node as well.

If you instead mean the internal robot state returned by `GetRobotState.srv`, the service is currently the only provided way to get this information using ROS. If it is necessary for your application, you can edit the hardware interface and register a custom callback directly with DRFL to monitor state changes (similar to what is done [here](https://github.com/DoosanRobotics/doosan-robot2/blob/humble/dsr_hardware2/src/dsr_hw_interface2.cpp#L169-L177)) and publish this information to a topic.

As the original issue seems to be fixed, I am closing this thread. If you re-encounter this issue, please create a new bug report or request this one to be reopened. Thank you.

closed 2025-12-02 by yurirocha15