DoosanRobotics/doosan-robot2 #287: Topic for Robot State

gauthammsam57, 2025-09-18:
I see that there is a topic related to the Robot State in the documentation, but I am unable to find it in an active state.

Is there a way to access the force and current pose of the TCP live, preferably through topics?

[image]

Namsangee, 2025-09-19:
Thank you for your question. To better assist you, could you please share which version you are currently using? Additionally, if you have any related logs available, it would be very helpful if you could provide them.

Namsangee, 2025-09-22:
Thank you for your question. Although the RobotState topic is defined in the documentation, it is expected that this topic is not published in the current runtime environment.
In the default bringup of dsr_bringup2, only /dsr01/joint_states and control-related services are active. The RobotState topic requires a separate publisher node to be enabled before it can be subscribed to.

That said, you can still access the TCP pose and force/torque information through the following methods:

TCP Pose

- By using the /dsr01/joint_states topic together with the robot’s URDF, you can perform forward kinematics to obtain the current TCP pose.

- In ROS, this computation is typically handled automatically, and you can directly subscribe to the /tf tree to get the transform between the base link and the TCP (tool) frame.

TCP Force/Torque

- You can call the /dsr01/aux_control/get_tool_force service to retrieve the force and torque applied to the TCP.

gauthammsam57, 2025-09-22:
Thanks for the responds @Namsangee.

I got it regarding /tf_tree. Will try to proceed with that for pos.

Regarding TCP force; basically I want to build force based pick and place. To be able get the force in real time would be the best bet, if we use service it would be very slow, right? 

Do you have any suggestions to get the data live? Or any other alternative to get this task done?