DoosanRobotics/doosan-robot2 #288: 로봇 제어권 관련 문의

jeong202017, 2025-09-19:
현재 bringup을 이용하여 로봇 제어를 할 경우

로봇의 제어권이 pc로만 넘어가 핸드 티칭 모드로 접근이 불가한 것으로 알고 있습니다.

이와 관련하여 pc와 핸드 티칭이 동시에 가능한 방안이 존재한다면 어떻게 사용하는지 알 수 있을까요?

Namsangee, 2025-09-19:
bringup(외부 제어) 상태에서도 Hand Teaching(Direct Teaching)은 사용 가능합니다.
ROS 노드/연결을 끊을 필요는 없으며, 모션이 실행 중이 아닐 때 티치펜던트에서 Direct Teaching을 활성화(또는 플랜지의 Direct Teach 버튼을 누른 상태로)하면 즉시 핸드티칭 모드로 전환됩니다. 핸드티칭 중에는 PC에서 보낸 이동 명령은 보류/거부되며, 버튼을 놓거나 Direct Teaching을 종료하면 바로 PC 제어로 복귀합니다.

jeong202017, 2025-09-19:
> bringup(외부 제어) 상태에서도 Hand Teaching(Direct Teaching)은 사용 가능합니다.
> ROS 노드/연결을 끊을 필요는 없으며, 모션이 실행 중이 아닐 때 티치펜던트에서 Direct Teaching을 활성화(또는 플랜지의 Direct Teach 버튼을 누른 상태로)하면 즉시 핸드티칭 모드로 전환됩니다. 핸드티칭 중에는 PC에서 보낸 이동 명령은 보류/거부되며, 버튼을 놓거나 Direct Teaching을 종료하면 바로 PC 제어로 복귀합니다.

https://github.com/user-attachments/assets/8e380162-f605-43fc-b347-eab8ac632bd5

song-ms, 2025-09-19:
@jeong202017 님 

안녕하세요 핸드가이딩을 하기 위해서는 로봇 모드가 자동 모드가 아닌 수동모드 상태에서만 가능합니다.
set_robot_mode(manual) 으로 설정하셔서 시도하시면 정상적으로 동작할 것으로 보입니다.

++ @Namsangee

jeong202017, 2025-09-19:
> [@jeong202017](https://github.com/jeong202017) 님
> 
> 안녕하세요 핸드가이딩을 하기 위해서는 로봇 모드가 자동 모드가 아닌 수동모드 상태에서만 가능합니다. set_robot_mode(manual) 으로 설정하셔서 시도하시면 정상적으로 동작할 것으로 보입니다.
> 
> ++ [@Namsangee](https://github.com/Namsangee)

덕분에 해결했습니다!! 감사합니다

closed 2025-09-19 by jeong202017